ZigEmbeddedGroup/microzig #42: Implement compiler_rt for AVR

mattnite, 2022-04-01:
This work requires upstreaming to https://github.com/ziglang/zig

As a repro compile this program with the following build command:

```zig
export fn foo(a: i32, b: i32) i32 {
    return a * b;
}
```

```
zig build-exe -target avr-freestanding -mcpu=avr5 test.zig -O ReleaseSmall
```

and you'll see the following output:

```
LLVM ERROR: Not supported instr: <MCInst 312 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:53>>
```

note: doing this in a basic program that is being built by the microzig build function, it does not bundle compiler_rt, so you'll see a missing definition:

```
ld.lld: error: undefined symbol: __mulsi3
```

MasterQ32, 2022-04-03:
gcc reference code:
https://github.com/gcc-mirror/gcc/blob/master/libgcc/config/avr/lib1funcs.S

kassane, 2022-11-11:
- https://github.com/ziglang/zig/issues/10372

oliverpool, 2023-09-09:
Still failing on zig 0.11.0, but with another error apparently:

```
> zig build-exe -target avr-freestanding -mcpu=avr5 test.zig -O ReleaseSmall
LLVM Emit Object... LLVM ERROR: Expected a constant shift amount!
[1]    14849 IOT instruction (core dumped)  zig build-exe -target avr-freestanding -mcpu=avr5 test.zig -O ReleaseSmall
```

kanashimia, 2023-09-13:
@oliverpool That is a bug in the LLVM 16, updating to LLVM 17 should solve that, Zig has llvm17 branch if someone wants to check.

oliverpool, 2023-09-13:
@kanashimia good to know, thank you!

According to https://llvm.org/ version 17.0.0-final should be released next week (rc4 was published last week) :hourglass:

oliverpool, 2023-09-14:
_Just documenting my failures until now, maybe someone can make sense of it_

I am trying to build the following program for Aduino Uno using https://github.com/ZigEmbeddedGroup/microchip-atmega
```zig
const micro = @import("microzig");

pub fn main() void {
    const led = micro.core.experimental.gpio.Gpio(micro.core.experimental.Pin("PB5"), .{
        .mode = .output,
        .initial_state = .low,
    });
    led.init();

    while (true) {
        busyloop();
        led.toggle();
    }
}

fn busyloop() void {
    const limit = 100_000;

    var i: u24 = 0;
    while (i < limit) : (i += 1) {
        @import("std").mem.doNotOptimizeAway(i);
    }
}
```

I have compiled the `llvm17` branch of https://github.com/ziglang/zig-bootstrap/tree/llvm17

```
zig version
0.12.0-dev.203+d3bc1cfc4
```

---

Let's run `zig build  -Doptimize=ReleaseSmall` for the first error: 
```
deps/microzig/src/modules/cpus/avr5.zig:106:9: error: runtime @panic not allowed in naked function
        @panic("Unhandled interrupt");
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
deps/microzig/src/modules/cpus/avr5.zig:115:25: error: runtime call not allowed in naked function
        copy_data_to_ram();
        ~~~~~~~~~~~~~~~~^~
```

I worked around by removing `callconv(.Naked)` from `microzig_unhandled_vector` and `microzig_start` (this is probably a very bad way to solve it, but I just want to compile something for now :see_no_evil: )

---

Next error:
```
deps/microzig/src/core/experimental/pin.zig:29:12: error: root struct of file 'ATmega328P' has no member named 'parse_pin'
```

I replaced `parsePin` with `parse_pin` in 
https://github.com/ZigEmbeddedGroup/microchip-atmega/blob/081683a4518f173e3ce7e7886fbbf39939ce4aaf/src/hals/ATmega328P.zig#L19

Cc @MasterQ32 : it seems that #103 changed some names. Shall I make a PR against microchip-atmega to attempt fixing some of them?

---

Next error:
```
src/hals/ATmega328P.zig:63:50: error: root struct of file 'microzig' has no member named 'gpio'
```

I replaced `micro.gpio.State` with `micro.core.experimental.gpio.State` in https://github.com/ZigEmbeddedGroup/microchip-atmega/blob/081683a4518f173e3ce7e7886fbbf39939ce4aaf/src/hals/ATmega328P.zig

---

Next error:
```
zig build-exe Arduino Uno.minimal ReleaseSmall avr-freestanding-eabi: error: error: couldn't allocate input reg for constraint 'r'
```

I have no idea what to tweak to get past this error... So I will leave it there for now :smile:

oliverpool, 2024-01-31:
Seems to be fixable with 0.12.0-dev, according to https://github.com/FireFox317/avr-arduino-zig/pull/8