ZigEmbeddedGroup/microzig #72: Expand EmbeddedExecutable

MasterQ32, 2022-07-27:
Expose common functions like `install()` and `setBuildMode()`, also `installRaw()`

kuon, 2022-08-02:
I also use:

```
addCSourceFile
addIncludeDir
addSystemIncludeDir
want_lto
```

I currently just use `exe.inner` I don't know if we need to wrap all of those.

r4gus, 2022-09-14:
Implemented in #77

kuon, 2022-09-14:
I just had to do the following:

```
    var buf: [100]u8 = .{0} ** 100;
    const date = std.fmt.bufPrint(
        &buf,
        "{d}",
        .{std.time.timestamp()},
    ) catch unreachable;
    const options = b.addOptions();
    options.addOption([]const u8, "date", date);
    exe.inner.addOptions("compile_info", options);

    exe.addPackage(.{ .name = "compile_info", .source = options.getSource() });

```

To be able to do

```
const date = @import("compile_info").date;
```

So I think #77 should be extended with `addOptions` and the code should look like:

```
    var buf: [100]u8 = .{0} ** 100;
    const date = std.fmt.bufPrint(
        &buf,
        "{d}",
        .{std.time.timestamp()},
    ) catch unreachable;
    const options = b.addOptions();
    options.addOption([]const u8, "date", date);

    exe.addOptions("compile_info", options); // this should be implemented

    // addPackage should not be needed

```

mattnite, 2025-02-16:
Closing because the build API has been refactored

closed 2025-02-16 by mattnite