ZigEmbeddedGroup/microzig #92: [Tracking issue] Platform support

enderger, 2022-12-02:
Currently, the number of supported platforms is rather minimal. If the owners are alright with it, I'd like to designate this issue as a tracking issue for boards/chips which are not yet implemented in here with the aim of getting a good selection supported.

To start off the list, here's a few chips I have come across:
- [ ] ATMega644/A (chip)
- [ ] ATTiny25/45/85 (chip)

MasterQ32, 2022-12-03:
Those two shouldn't be a problem to bring up, but i'd wait for the `regz` rewrite to complete, so we can easily generate just *all* ATtiny and ATmega descriptors/register files and ship them in a `avr` package for microzig

MasterQ32, 2022-12-03:
Ah but: We can make this issue a good tracking issue for platforms people want to see!

sogwms, 2023-05-14:
please support vexpress-a9 (on qemu)

Khitiara, 2023-05-14:
is the 6502 applicable to microzig or is that more suitable for general zig rather than here?

MasterQ32, 2023-05-14:
> is the 6502 applicable to microzig or is that more suitable for general zig rather than here?

See https://github.com/ziglang/zig/issues/6502. Right now, Zig cannot compile to the 6502, so we can't target it either. MicroZig is a framework and library for Zig, not another programming language. Also 6502 isn't an embedded target, more a regular "PC CPU" in that regard

GustavoSchip, 2024-04-20:
Is there a way to support the Arduino Nano RP2040 Connect?
It does have a load of peripherals but the printout of the board is different than the standard Pico

MasterQ32, 2024-04-21:
> Is there a way to support the Arduino Nano RP2040 Connect? It does have loads of peripherals but the pinout of the board is different from the standard Pico. If the correct pins and their features can be used that would be awesome!

You can always use a Pico as a base board, as long as the flash and clock does match. Apart from that, GPIOs are always numbered GPIO0 to GPIO30, so you can just look them up in the schematic :)

GustavoSchip, 2024-04-21:
> You can always use a Pico as a base board, as long as the flash and clock does match. Apart from that, GPIOs are always numbered GPIO0 to GPIO30, so you can just look them up in the schematic :)

I assume that the clock is the same, it would make sense from a compatibility perspective. Il look into it, thanks! 👍 

Just labelling pins without restricting what it does is indeed the more versatile approach, that way a wide variety of RP2040 boards are supported! 🎉

Khitiara, 2024-08-16:
Im interested in AT91SAM7S256 as a chip to be supported for use in making firmware for LEGO mindstorms NXT; is that one usable with some sort of config tweaks or combinations or would it need dedicated support?

gajwani, 2024-08-31:
Raspberry Pi released the new [Pico 2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/) based on the [RP2350](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf) microcontroller, that is the successor to the RP2040.

I'm looking in to providing a PR for the BSP (I'm new to both the microcontroller, and Zig and the RP2350 has both ARM and RISC-V cores), but I wanted to register interest, in case anyone else is working on it, or wants to collaborate.

ilioscio, 2024-12-16:
[Teensy](https://www.pjrc.com/teensy/) microcontroller support would be awesome (especially the newest 4.0/4.1 versions), I would love to do embedded audio in native zig, it seems like [this repo](https://github.com/tsunko/teensy-minimal-bare-metal-zig) has something of a working baremetal example but I am far too ignorant to know if any of that work integrates with this project in any capacity.