ZigEmbeddedGroup/microzig #119: documentation not clear and failre when using example

jmgurney, 2023-04-01:
I'm attempting to use microzig, and I'm also very new to zig and the instructions in `README.adoc` assume that you have a lot of knowledge about zig and how microzig works.

I attempted to use the sample and despite inferring how to make it work, even with modifications, it fails to compile.

run: `zig init-exe`

undocumented commands (implied via the vague statement: `add the hardware support package as a submodule`, this is also the "simple way" as `zig init-exe` doesn't do a `git init`, so can't do some `git submodule` command to add it):
```
mkdir deps
cd deps
git clone --recursive https://github.com/ZigEmbeddedGroup/microchip-atmega
cd ..
```

replace `build.zig` (undocumented what file, and need to replace `hardware_support_package` with `microchip-atmega` and replace `root_source_file` with `source_file`) w/:
```
const std = @import("std");
const atmega = @import("deps/microchip-atmega/build.zig");

// the hardware support package should have microzig as a dependency
const microzig = @import("deps/microchip-atmega/deps/microzig/build.zig");

pub fn build(b: *std.build.Builder) !void {
    const optimize = b.standardOptimizeOption(.{});
    var exe = microzig.addEmbeddedExecutable( b, .{
        .name = "my-executable",
        .source_file = .{
            .path = "src/main.zig",
        },
        .backing = .{
            .board = atmega.boards.arduino_nano,

            // instead of a board, you can use the raw chip as well
            // .chip = atmega.chips.atmega328p,
        },
        .optimize = optimize,
    });
    exe.install();
}
```

replace `src/main.zig` with the next block (undocumented what file to put it in).

run `zig build` and see the following failure:
```
$ zig build
zig build-exe my-executable Debug avr-freestanding-eabi: error: the following command terminated unexpectedly:
[xxx]/github/zig/build/zig2 build-exe /private/tmp/mcre/deps/microchip-atmega/deps/microzig/src/start.zig -fno-strip --cache-dir /private/tmp/mcre/zig-cache --global-cache-dir [xxx]/.cache/zig --name my-executable -fno-compiler-rt -fsingle-threaded -target avr-freestanding-eabi -mcpu avr5 --script /private/tmp/mcre/zig-cache/microzig/bad2341405a5fe8c06f6286f91ea1100/ATmega328P_AVR5.ld --mod microzig:config,chip,cpu,hal,board:/private/tmp/mcre/deps/microchip-atmega/deps/microzig/src/microzig.zig --mod chip:microzig:/private/tmp/mcre/deps/microchip-atmega/src/chips/ATmega328P.zig --mod board:microzig:/private/tmp/mcre/deps/microchip-atmega/src/boards/arduino_nano.zig --mod hal:microzig:/private/tmp/mcre/deps/microchip-atmega/src/hals/ATmega328P.zig --mod app:microzig:/private/tmp/mcre/src/main.zig --mod cpu:microzig:/private/tmp/mcre/deps/microchip-atmega/deps/microzig/src/modules/cpus/avr5.zig --mod config::/private/tmp/mcre/zig-cache/microzig/config-6df6165a62f30831cad66fa94379f8d3.zig --deps app,microzig --enable-cache --listen=- 
Build Summary: 1/4 steps succeeded; 1 failed (disable with -fno-summary)
install transitive failure
└─ install my-executable transitive failure
   └─ zig build-exe my-executable Debug avr-freestanding-eabi failure
      └─ linkerscript success
error: the following build command failed with exit code 1:
/private/tmp/mcre/zig-cache/o/b840765e239219fa1562b810d024cd97/build [xxx]/github/zig/build/zig2 /private/tmp/mcre /private/tmp/mcre/zig-cache [xxx]/.cache/zig
```

environment: MacOS Ventura 13.2

zig version: `0.11.0-dev.2249+dcdb87836`

I'm unsure what to do now to make the necessary executable.

oliverpool, 2023-09-09:
I am facing the same issue. Just building the latest https://github.com/ZigEmbeddedGroup/microchip-atmega fails as well: https://github.com/ZigEmbeddedGroup/microchip-atmega/issues/25

Did you find a workaround?

mattnite, 2025-02-16:
Hey @jmgurney sorry we never got around to this. We've overhauled the build API and documentation so I'm closing this issue for now.

closed 2025-02-16 by mattnite