ZigEmbeddedGroup/microzig #161: Support serial over USB (CEC ADC)

negz, 2023-12-11:
I'd like to have a log.zig logFn that I could use to emit debug logs over USB, rather than UART.

Currently Microzig has support for acting as a USB HID device. I believe emitting debug logs over USB would require support for acting as a "serial device", which uses USB CDC/ADC.

negz, 2023-12-11:
https://github.com/tinygo-org/tinygo/blob/731532cd2/src/machine/usb.go#L261

I'm attempting to take a shot at this, using TinyGo's implementation as a reference. In part because it seems more minimal than alternatives like TinyUSB. I believe TinyGo have implemented just enough of CEC to be a serial device, not support for modems etc. I'm also trying to use TinyGo as a reference because I'm fluent in Go, but not Zig, or microcontrollers, or USB. 😄

rosshadden, 2024-01-31:
Any luck with this? I read that picotool needs the firmware running on a microcontroller to use USB-CDC in order for its `load --force` argument to work (letting you reboot picos outside of BOOTSEL mode).

Apparently this is because picotool uses [PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L80) though I don't know enough of how this stuff works to provide more insight.

negz, 2024-01-31:
Unfortunately I didn't get very far before my vacation ended.

I was finding it pretty hard to figure out how to cleanly extend the existing USB implementation. A _lot_ of this is probably just because I don't know Zig, but I think it was exacerbated by the fact that the implementation isn't designed for extensibility.

My understanding based on https://github.com/ZigEmbeddedGroup/raspberrypi-rp2040/pull/40 is that the USB implementation is a port of https://github.com/cbiffle/rp2040-usb-device-in-one-file/blob/4878f4/src/main.rs, which per its header comments is intended "for reference and study".

Here's what I ended up with. It's mostly just all the structs that I think are needed to make CDC work: https://github.com/ZigEmbeddedGroup/microzig/commit/846e994e

arkadiuszwojcik, 2024-06-02:
I have some progress on this, but further work depends on this PR: #201. Chunked data support is required because USB CEC ADC config descriptor length is above regular packet size (64 bytes).

arkadiuszwojcik, 2024-06-11:
With #201 merged, I was playing around with USB-CDC. I had a working prototype, but the way the config was defined was very ugly. Now I have a new prototype with a much nicer approach. I would like to know your opinion on it. Here is main [file](https://github.com/arkadiuszwojcik/microzig/blob/usb_cdc/examples/raspberrypi/rp2040/src/usb_cdc.zig) and [here](https://github.com/arkadiuszwojcik/microzig/blob/usb_cdc/core/src/core/usb.zig#L554) is the main tagged union that does the trick with some comptime magic. Config code can be simplified more with default values for some fields. There is also new `serialize_buff` (temporary name) method and ole one `serialize` left for comparison. Any pros and cons of such solution?

@r4gus  @negz @rosshadden