ZigEmbeddedGroup/microzig #169: Generate status matrix

MasterQ32, 2024-02-17:
We had a nice support matrix on the [ZEG Front Page](https://github.com/ZigEmbeddedGroup), but it was hand-maintained. This didn't really scale well, so the idea is that the `tools/bundle.py` also creates a file https://downloads.microzig.tech/.data/bsp-overview.svg that shows a table of all board support packages and their current status

coolaj86, 2025-02-27:
Any update on this?

What does MicroZig support?

Which boards are best supported?

Do any have WiFi or BLE drivers or driver-wrappers?

mattnite, 2025-03-02:
Hey @coolaj86, we've had some great improvements to the build script that gets us closer to achieving this but it's not there yet.

Right now RP2040 and RP2350 are the best supported, followed by ESP I believe, then STM32.

If you take a look at the driver framework you'll see a number of MCU agnostic drivers for devices.

I'm going to close this ticket not because we've done this, but because it doesn't reflect what we're going to do.

closed 2025-03-02 by mattnite