ZigEmbeddedGroup/microzig #171: Various issues

Grazfather, 2024-02-17:
I am having troubles getting examples building with the monorepo. Reporting my issues as requested.

1. _bundle.py_ uses `pathspec` with the `negate` kwarg in `match_tree`, but _requirements.txt_ doesn't pin a version. This used 0.11.0 on my machine, but `negate` shows up in 0.12.0
2. _bundle.py_ uses the `--hard-deference` option from `tar`, which is not a valid option for OSX `tar`. I suggest using `tarlib`. I got around this, however, by installing `gnutar` with `brew` and symlinking it to `tar` on my `PATH`.
3. No _.zon_ files with the _build.zig_ files for the example. I don't know how to write these to have it point to the server.

MasterQ32, 2024-02-18:
> No .zon files with the build.zig files for the example. I don't know how to write these to have it point to the server.

I will add this in more detail to the readme, but for now:
Copy/unpack the example base from `deployment-zig`, which will have the correct `build.zig.zon`, and you can then use that repo as a testing place

Grazfather, 2024-02-18:
They are much nicer now! I was able to build the rpi2040 examples when following them again.

Grazfather, 2025-03-13:
I guess I have been playing with this repo for over a year now!

These issues are stale.

closed 2025-03-13 by Grazfather