ZigEmbeddedGroup/microzig #191: ATmega328P SRAM memory map

CrispyCrafter, 2024-05-14:
Hi all I'm new to the project as studying the source code to get a better handle on how MicroZig works

I have a question regarding the `.memory_regions` specified in https://github.com/ZigEmbeddedGroup/microzig/blob/ea809c6b7f18583042cd28e3fee6eb11510c0ad3/bsp/microchip/avr/build.zig#L26-L27

To help me get a better understanding I have pulled the `ATmega328P.atdf` directly in conjunction to the [hardware reference manual](https://github.com/eerimoq/hardware-reference/blob/master/Atmel/atmega328p%20reference%20manual.pdf) 

Based on this the memory address starts at `0x000000` with the following block sizes
| Sector      | Size   | Offset  |
|-------------|--------|---------|
| prog        | 0x8000 | 0x00000 |
| signatures  | 0x0003 | 0x08000 |
| fuses       | 0x0003 | 0x08003 |
| lockbits    | 0x0001 | 0x08006 |
| data.ram    | 0x0100 | 0x08007 |

I'm not sure how to get to the ram offset `0x800100` as set here:
```
.memory_regions = &.{
    .{ .offset = 0x000000, .length = 32 * 1024, .kind = .flash },
    .{ .offset = 0x800100, .length = 2048, .kind = .ram },
},
``` 

My end-game here is to add support for `AVR8 - tinyAVR` - specifically the Attiny85

MasterQ32, 2024-05-14:
> I'm not sure how to get to the ram offset 0x800100 as set here:

Reading the docs for `avr-gcc` and hoping that LLVM will do the same, sadly :(

basically for "RAM" offset everything by `0x800000`

CrispyCrafter, 2024-05-15:
Well that's quite something. Thanks for putting me on the right trail

closed 2024-05-15 by CrispyCrafter