ZigEmbeddedGroup/microzig #208: HAL Design

ikskuh, 2024-06-28:
I created a [Google Doc](https://docs.google.com/spreadsheets/d/1ilo_q4Qu_hXvhlKXskjjVudeR-lz2zwig7RFBfjtiNU/edit?usp=sharing) which contains some overview of what features different MCUs have. 

Those who want to join can send me a message with their email on discord, i'll invite you

haydenridd, 2024-06-28:
Alright! So, apologies in advance for the dissertation on HALs but... I've been thinking about HAL's for a very long time in my embedded career. I've used a bunch of them, seen even more, and even wrote my own internal HALs for projects. I also feel like I have what may be a somewhat controversial opinion, so I want to make sure to fully articulate my opinion and my reasoning behind it :)

# High Level Summary
- I think "universal HALs" are a trap that should be a avoided
    - This is not to say HALs can't be written that abstract multiple chips, or even multiple chip families, but it's all dependent on the homogeneity of the MCUs they cover
    - This is the big one that may generate some disagreement, so please read "The Case Against Universal HALs" below before flaming please!
 - A couple of fully fleshed out, highly reliable, specific HALs for the most popular chip families (rp2040, arduino, ST?) will do us more benefit than trying to "drink the sea" creating the "perfect API" that covers all possible chips
 - Meeting developers where they're at with multiple choices for abstraction level will help us cast a wider net
 - We shouldn't be afraid to continuously try something, throw it away if we don't like it, repeat. It's okay to not get it perfect on our first go!
 
 ## The Case Against Universal HALs
 
 ### Definition
 
 First, let me define what I mean by "Universal HAL". In my mind, a universal HAL is one that attempts to provide a hardware abstraction that works across many (if not all) MCUs in given ecosystem. A very good example of the closest thing "in production" to a universal HAL I can think of would be the [zephyr project](https://zephyrproject.org/). Zephyr supports a very, very large number of MCUs, and ignoring some of the issues with this approach, abstracts their various peripherals using a "device tree" approach similar to Linux. 
 
### Problem Statement
Now, let me make a bold statement:
Universal HALs attempt to solve a problem that doesn't actually exist.
 
"But", you say, "there's plenty of good reasons to want to compile my firmware for multiple targets effortlessly!" And you would be absolutely right! The issue with a universal HAL is it isn't just trying to let you use the same API for two or three targets, it's trying to let you use it for _all_ targets. Think about how often you truly need to compile your firmware for 2 different MCUs. I can think of some times in my career, usually due to supplier shortages, or experiments. Now think about how often you compile your firmware for 5+ different MCUs... Okay I can maybe think of some hypothetical cases but they're starting to get incredibly niche. Okay now 20+... You see where I'm going. With how different peripherals, pin counts, and feature sets can be MCU to MCU, the amount of application logic you can "share" between chips starts to drop off rapidly. 
 
But if a universal HAL is feasible, what's the big deal? Who cares if you don't _need_ to compile for every supported chip, isn't it nice to have a standard API anyways? Well, that brings me to...
 
 ### The Cost of Abstraction

 I love abstraction. I use it constantly, we all do. It's why we write in C instead of assembly, or C++ instead of C, or MicroPython instead of C++. Each level of abstraction brings us away from processor instructions and memory mapped IO, and towards more digestible concepts like GPIOs being "on" or "off", and reading a voltage in millivolts from an ADC as opposed to a 16 bit signed integer. 
 
I like to think of each layer of abstraction as a contract. Using C, if you follow the rules regarding the language and the compiler settings, the contract says that it will produce valid assembly instructions for your processor of choice. Zig + LLVM can be thought of as having a similar contract for compiling code for a given CPU. So what contract does a HAL provide? Well, generally speaking the contract is: "You may not have the flexibility and customization that comes with flipping bits in registers, but if you use this your code can convey _behavior_ rather than setting bits in registers".

And this is a nice trade, with the ever growing complexity of MCUs, it would be a lot to ask someone to start every project from scratch, transcribing all the peripheral addresses into code. But what happens when the contract doesn't hold? Well, you essentially forfeit that layer of abstraction, and have to learn the layer `N-1` down. Here's an example function from an STM32 HAL:
``` C
void HAL_GPIO_WritePin(GPIO_TypeDef* GPIOx, uint16_t GPIO_Pin, GPIO_PinState PinState)
{
  /* Check the parameters */
  assert_param(IS_GPIO_PIN(GPIO_Pin));
  assert_param(IS_GPIO_PIN_ACTION(PinState));

  if(PinState != GPIO_PIN_RESET)
  {
    GPIOx->BSRR = GPIO_Pin;
  }
  else
  {
    GPIOx->BSRR = (uint32_t)GPIO_Pin << 16;
  }
}
```
Not too bad, right? There's some stuff we don't know off hand but it's relatively clear we're setting some register we can probably look up in the datasheet by name. If this fails, this isn't too awful a level to have to figure out. 

Now let's look at the same example from Zephyr (taken directly from the [example application](https://github.com/zephyrproject-rtos/example-application/blob/main/app/src/main.c)):
``` C
ret = blink_off(blink);
if (ret < 0) {
    LOG_ERR("Could not turn off LED (%d)", ret);
    return 0;
}
```

What do we have to look at if this call fails? Well, let's see:

Def of `blink_off`:
``` C
[... cut ...]
- [cycle repeats, support more and more chips/board as Zig marches towards 1.0.0]
- Zig 1.0.0 releases
- Eventually Zig becomes popular enough MCU vendors are noticing, we start offering to help integrate Zig's compiler/build system with their vendor code/toolchains
- ZIg has achieved market supremacy, _vendors_ are now writing their HALs in Zig instead of C
- Sipping mojitos on beach, watch sun set slowly, screen fades to black

Thank you for taking the time to listen to my thoughts/rant, and I'm looking forward to the discussion!

- Hayden

DNedic, 2024-06-29:
First off, I do think a HAL with an universal API is a good idea, but it should not attempt to wrap everything in my opinion. Some things can be left out of the HAL and remain port-specific in order to avoid either limiting what the user can do with `microzig`, or horribly hacky solutions to adapt to an overly simplified API.

To address some of the points @haydenridd made:

> "But", you say, "there's plenty of good reasons to want to compile my firmware for multiple targets effortlessly!" And you would be absolutely right! The issue with a universal HAL is it isn't just [...]

Hard disagree here. While a single peson might not need this, it enables the ecosystem to grow as now reusable libraries can be written. One major reason Arduino took off is because of the availability of libraries, and those wouldn't be there without an universal HAL. Rust is seeing the same ecosystem bloom effect with its `embedded-hal`.

On the cost of abstraction: I tend to agree to an extent, this is something to be careful about and too many layers of indirection can backfire. On the other hand with LTO the runtime cost should be near 0 most of the time.

> Write a HAL for each family of chips that is appropriately abstractable

This is what is being done now to a degree and I don't see much traction behind the embedded Zig ecosystem. Also it's a huge duplication of work, and a deterrent to anyone who wants to add chip support to `microzig`. I have been considering writing a HAL for Espressif chips however I am much less inclined if I know it's not going to be a part of a standardized API on top of which libraries, educational content etc can be written.

> Let users "opt-in" to the level of abstraction they would like

A HAL is almost always opt-in, and you should always be able to use lower layers as well as use chip/vendor specific functionality on the side as well. Having a standardized API should not stop nonstandard extensions.

haydenridd, 2024-06-29:
> First off, I do think a HAL with an universal API is a good idea, but it should not attempt to wrap everything in my opinion. Some things can be left out of the HAL and remain port-specific in order [...]

I'll do a little walking back and say I agree with you here. I think there _is_ value to a universal "simple" API, and then a per-family more specialized API. I would just emphasize they should be separate, and we shouldn't bend over backwards to try to fit everything under one "master" API. 

I realize I'm coming at this from my own perspective of a professional firmware engineer, so there are absolutely biases in what I want and what is _useful to me_ professionally. Arduino is a great example to look at. It is something that is wildly successful, great in bringing people into the community, and _useless for me professionally_. I can't write firmware in Arduino, it doesn't get low level enough. It's still extremely valuable to have in the ecosystem, it's just not the right tool for me. It just means there needs to be a _seperate_ tool (C with vendor HALs) that gets low level enough.

I think my fear (again putting on my professional firmware engineer hat), is that we never get past the "basic API" level. I've seen this a lot in the Rust embedded scene. You excitedly download a HAL crate for your chip only to realize it more or less only supports some basic GPIO, blocking UART reads, etc. It's frustrating because it puts you in an odd position. You can:
- Contribute to the project to try to make it more expansive (the best answer, but requires you learning the existing code-base and essentially writing a HAL from scratch)
- Mix existing basic HAL with raw register writes
- Sadly walk back to your abusive partner C/C++ 

I love that Rust is gaining more traction in embedded, but there is only a single company I know of using it in production for deeply embedded (freestanding) MCUs:
https://github.com/oxidecomputer/hubris

And they basically wrote an OS from scratch and don't really use any of the community HALs (they do use the cortex-m related crates but no higher level abstractions than that as far as I can tell). 

Maybe a better way to frame my (selfish) fear is that sometimes I think the universal, easy API gets prioritized to the point where the nitty-gritty is somewhat forgotten about. So then you have tons and tons of HALs that get you to blinky and some basic peripheral usage, and not much else. Maybe a point to take from this is I want us to be very clear in our intentions and what is supported at what time. Similar to "tierN" support targets, being transparent that:
- These chips are fully fleshed out
- These chips can blink an LED
- We don't even know what these chips are

Finally to address this point:
>This is what is being done now to a degree and I don't see much traction behind the embedded Zig ecosystem. Also it's a huge duplication of work...

I hear you here. I think I envisioned any of these "family specific" HALs still falling under an "official" ZEG Github project (or potentially mono-repo).  I think a single source of truth is super important, and yes, I feel the pain of wondering "is this project just going to get lost in the sands of time or actually used?". Going back to your first point, I like the idea of a "core" API for simple stuff, and "extensions" for vendor specific peripherals. But they should all be under one roof, and it should be obvious what you should use when "starting an embedded project with Zig for X". A "core" API for the simple stuff should take care of most of the duplication, once you start getting into specific peripherals I don't think there's a ton of code re-usabilty between different vendors. 

I appreciate the opposing perspective :) Libraries _using MCU peripherals_ was not something I thought too much about but would certainly be a nice addition to the ecosystem.

taylorh140, 2025-01-19:
I have considered that HAL might be an abstraction for the sake of abstraction. After working with microzig a bit (specifically on the STM32 stuff) at least in that works it might be nicer to build specific code for handling specific "peripheral chiplets?" (timer_v1.TIM_G16 etc)  as apposed to generating a HAL. It's noteworthy that each peripheral has its own unique capabilities that can be targeted, but having code that targets the best functionality of these will leave space for "abstract behavior" to bubble to the top. 

Of course, I'm not opposed to HAL for very simple devices like UARTs.  but I pretty much never use these due to the typical blocking implementation.

curuvar, 2025-04-12:
Throwing in my two cents -- one of the strengths of a HAL is that you don't have to remember all the fiddly-bits of multiple different processors to do simple things like turn a GPIO pin on or off or have a timer call a function. I appreciate that, and I would expect there to be a common microzig way of doing them. 

Beyond that, there should be a common theme for handling the more complex things.  Even when things cannot easily be abstracted perfectly, if someone becomes familiar with microzig on one chip, moving to another should seem familiar.