ZigEmbeddedGroup/microzig #210: error: struct 'chip.devices.RP2040.peripherals' has no member named 'NVIC'

tanj, 2024-06-29:
I'm trying to play around with the synth project from the 2023 sycl conference and I'm getting a build error when trying to use the latest microzig version. I've made some changes to the build script to try get it working, but I'm hitting an error now that seems to me to be related to microzig.

(I ran into a hash issue using the vanilla project deps which is why I'm trying to use the latest microzig).

```
% /c/src/zig/zig-windows-x86_64-0.12.0/zig.exe build
install
└─ install generated to uart_monitor.elf
   └─ zig build-exe uart_monitor Debug thumb-freestanding-eabi 2 errors
C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\start.zig:70:81: error: expected type 'start.Options', found 'type'
pub const microzig_options: Options = if (@hasDecl(app, "microzig_options")) app.microzig_options else .{};
                                                                             ~~~^~~~~~~~~~~~~~~~~
C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\start.zig:46:21: note: struct declared here
pub const Options = struct {
                    ^~~~~~
referenced by:
    vector_table: C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\cpus\cortex_m.zig:151:50
    remaining reference traces hidden; use '-freference-trace' to see all reference traces
C:\Users\jtebokkel\AppData\Local\zig\p\1220a557b7c940ea0193e8e4f516c6dab6d16dd41e528486889f356a2388930aae42\src\hal\irq.zig:2:39: error: struct 'chip.devices.RP2040.peripherals' has no member named 'NVIC'
const NVIC = microzig.chip.peripherals.NVIC;
             ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
zig-cache\o\ce1fd5a7bdf3a724fe567c58a510dd57\chip.zig:66:33: note: struct declared here
        pub const peripherals = struct {
                                ^~~~~~
error: the following command failed with 2 compilation errors:
C:\src\zig\zig-windows-x86_64-0.12.0\zig.exe build-exe -fno-strip -ODebug -target thumb-freestanding-eabi -mcpu cortex_m0plus --dep app --dep microzig -Mroot=C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\start.zig --dep microzig --dep workshop -Mapp=C:\src\synth-workshop\demos\01_uart\monitor.zig --dep config --dep chip --dep cpu --dep hal --dep board -Mmicrozig=C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\microzig.zig --dep microzig -Mworkshop=C:\src\synth-workshop\src\workshop.zig -Mconfig=C:\src\synth-workshop\zig-cache\c\cb74838774d212d46a4b6ad47bca5bcc\options.zig --dep microzig -Mchip=C:\src\synth-workshop\zig-cache\o\ce1fd5a7bdf3a724fe567c58a510dd57\chip.zig --dep microzig -Mcpu=C:\Users\jtebokkel\AppData\Local\zig\p\1220a74a829bcd0e0cb6b5918646e20c68c2c47be9401e6f873a445453d4d8102027\src\cpus\cortex_m.zig --dep microzig -Mhal=C:\Users\jtebokkel\AppData\Local\zig\p\1220a557b7c940ea0193e8e4f516c6dab6d16dd41e528486889f356a2388930aae42\src\hal.zig --dep microzig --dep bootloader -Mboard=C:\Users\jtebokkel\AppData\Local\zig\p\1220a557b7c940ea0193e8e4f516c6dab6d16dd41e528486889f356a2388930aae42\src\boards\raspberry_pi_pico.zig -Mbootloader=C:\src\synth-workshop\zig-cache\o\730f41bc6b5f93cab44fdd7241cc9338\w25q080.bin --cache-dir C:\src\synth-workshop\zig-cache --global-cache-dir C:\Users\jtebokkel\AppData\Local\zig --name uart_monitor -static -fcompiler-rt --script C:\Users\jtebokkel\AppData\Local\zig\p\1220a557b7c940ea0193e8e4f516c6dab6d16dd41e528486889f356a2388930aae42\rp2040.ld --listen=-
Build Summary: 80/85 steps succeeded; 1 failed (disable with --summary none)
install transitive failure
├─ install generated to uart_monitor.uf2 transitive failure
│  └─ run elf2uf2 (uart_monitor.uf2) transitive failure
│     └─ zig build-exe uart_monitor Debug thumb-freestanding-eabi 2 errors
└─ install generated to uart_monitor.elf transitive failure
   └─ zig build-exe uart_monitor Debug thumb-freestanding-eabi (+9 more reused dependencies)
error: the following build command failed with exit code 1:
C:\src\synth-workshop\zig-cache\o\76617bd811a3831f1b1ec2d766d7f552\build.exe C:\src\zig\zig-windows-x86_64-0.12.0\zig.exe C:\src\synth-workshop C:\src\synth-workshop\zig-cache C:\Users\jtebokkel\AppData\Local\zig --seed 0x37e3cbaa -Z7c6f518096f843be
```

MatthiasPortzel, 2024-06-30:
Interrupt support regressed at some point (I wasn't a microzig user at the time but I think it was a result of switching to generating data from svd.) See #188 for how I hacked together interrupts.

Grazfather, 2025-03-25:
Covered by #188