ZigEmbeddedGroup/microzig #214: Usb.init_clk() freezes depending on timing

hryx, 2024-07-09:
I have been struggling to get USB working on any rp2040 board because the program always freezes in the call to `rp2040.usb.Usb.init_clk()`. However, @arkadiuszwojcik helped me debug tonight and found that it is possible to get it to work with a timing hack.

## Steps to reproduce

Modify the example `usb_hid.zig` around the clock initialization:

```zig
    ...
    std.log.debug("one", .{});
    // First we initialize the USB clock
    rp2040.usb.Usb.init_clk();
    std.log.debug("two", .{});
    ...
```

When flashing a Raspberry Pi Pico H or similar board, UART prints `one` but not `two`, and the LED never blinks.

Modify `bsp/raspberrypi/rp2040/src/hal/resets.zig` such that the empty while loop has at least one NOP instruction:

```zig
pub fn reset(mask: Mask) void {
    ...
    while ((RESETS.RESET_DONE.raw & raw_mask) != raw_mask) {
        asm volatile (
            \\ nop
        );
    }
}
```

...and suddenly, the LED blinks and we see `two` in the log. It also suffices to add some log lines inside `reset()` to change the timing of operations, although it may not be as consistent.

## Other notes

The stock `pico-examples/usb/device/dev_hid_composite` example from TinyUSB works out of the box.

arkadiuszwojcik, 2024-07-09:
Few lines below there is simillar fix:

```
// have to bitcast after a read() instead of `RESETS.RESET_DONE.raw` due to
// some optimization bug. While loops will not be optimzed away if the
// condition has side effects like dereferencing a volatile pointer.
// It seems that volatile is not propagating correctly.
while (@as(u32, @bitCast(RESETS.RESET_DONE.read())) & raw_mask != raw_mask) {}
```

I made some tests using such fix and also can confirm that without fix everytings works in `Debug` build, with fix it works also in `ReleaseSmall` build but doesn't work in `ReleaseFast` and `ReleaseSafe` builds.

hryx, 2024-07-10:
> `while (@as(u32, @bitCast(RESETS.RESET_DONE.read())) & raw_mask != raw_mask) {}`

Yep, this works for me as you described too. Another alternative suggested by a community member was `hw.tight_loop_contents();` inside the loop body, which also works.

PR #219 (2024-07-14): fix(rp2040) apply broken codegen workaround to both busy loop
And also remove couple of redundant `@as` usages.

Partially fixes #214
closed 2024-07-14 by ikskuh