ZigEmbeddedGroup/microzig #221: ESP support

DNedic, 2024-07-14:
This issue tracks the support status for ESP32 series Espressif SoCs and will be updated with new tasks continously:

- [ ] Add GPIO driver - [in progress](https://github.com/ZigEmbeddedGroup/microzig/pull/216)
- [ ] Add LEDC driver
- [ ] Add UART driver
- [ ] Add support for the ESP32-C6
- [ ] Add support for the ESP-IDF Application Image Format
- [ ] Enable caches

Grazfather, 2025-03-25:
#415 should knock off this one when it's ready.

> Add support for the ESP-IDF Application Image Format

Grazfather, 2025-04-11:
We also need an i2c driver, and most of these, to follow our standard hal api, will need a time driver so that we can use deadlines, so I think the time device should be next.

Uart was done by @tact1m4n3 in #436

tact1m4n3, 2025-04-11:
Interrupts can be checked 😄