ZigEmbeddedGroup/microzig #231: Example not working on rp2040

lMaxiCl, 2024-08-09:
I installed microzig as a dependency in my zig project using zig fetch {tarball from downloads.microzig.tech}, copypasted main.zig file from your examples folder for raspbery pi pico, copipasted build.zig file from examples root, deleted other build projects except blinky.

 When I run zig build:
 
actual result
---
error
**zig/blink/build.zig:2:26: error: no module named 'microzig/build' available within module root.@build
const MicroZig = @import("microzig/build");**

expected result:
---
**uf2 or elf file builds ready to flash on my pico board**

research:
---
 shows that in microzig build.zig.zon .build dependency not marked with @ so i believe this is a some sort of private dependency. I tried to change import to just ```@import( microzig)```, but then i receive error init method not found. 
 
 zig env(not full removed path variables)
  "target": "x86_64-linux.6.9.12...6.9.12-gnu.2.39",
 "env": {
  "ZIG_GLOBAL_CACHE_DIR": null,
  "ZIG_LOCAL_CACHE_DIR": null,
  "ZIG_LIB_DIR": null,
  "ZIG_LIBC": null,
  "ZIG_BUILD_RUNNER": null,
  "ZIG_VERBOSE_LINK": null,
  "ZIG_VERBOSE_CC": null,
  "ZIG_BTRFS_WORKAROUND": null,
  "ZIG_DEBUG_CMD": null,
  "CC": null,
  "NO_COLOR": null,
  "CLICOLOR_FORCE": null,
  "XDG_CACHE_HOME": null,

Thank you in advance for your help.

lMaxiCl, 2024-08-14:
In case someone is as dumb as I am, you can import microzig and use field build from it. But still tutorials should be fixed. I believe I can do it, but this will be my first commitment to open source project, so I better ask: should I change examples? Or you guys plan to fix it on microzig repo side?

Grazfather, 2025-03-13:
This is almost certainly fixed now. Please try building on an up-to-date main with zig 0.14.0.

To use microzig as a package, see how it's done [here](https://github.com/haydenridd/zig-rtt/blob/main/rp2040_example/src/main.zig#L3) and [here](https://github.com/haydenridd/zig-rtt/blob/main/rp2040_example/build.zig#L3)

closed 2025-03-13 by Grazfather