ZigEmbeddedGroup/microzig #250: Fully implement ST77xx driver

ikskuh, 2024-10-08:
This is a common display driver.

There's already some source code there, it's just not finished yet

mattnite, 2024-10-08:
@ikskuh could you please summarize what needs to be done so that someone could more easily pick this issue up?

ikskuh, 2024-10-09:
I'll do that when we have the basics figured out. I started doing the driver over a year ago and didn't do anything with it except updating the Zig version.

The idea was to basically port the initialization logic of the Arduino drivers here: <https://www.arduino.cc/reference/en/libraries/adafruit-st7735-and-st7789-library/>

But not copying their code structure, but only the raw command logic