ZigEmbeddedGroup/microzig #285: [ESP32-C3] How do I correctly setup the blink example on my own repository

JanBeelte, 2024-11-11:
Hello together,

I am pretty new to the zig build system and struggeling to setup ESP32-C3 support on my own repository using microzig as a dependency. Currently I have the following in my build.zig.zon file:
```
.{
    .name = "firmware/blink",
    .version = "0.0.0",
    .dependencies = .{
        .@"microzig/build" = .{  
            .url = "git+https://github.com/ZigEmbeddedGroup/microzig.git#d6a97c9476c3377877946873cad040f769e42698",
            .hash = "12204ade9b0ef08b6787667a99396aa1e89e8a6dca1e25b62b9490e3ab8394a0383c" },
        .@"microzig/port/espressif/esp" = .{ 
            .url = "git+https://github.com/ZigEmbeddedGroup/microzig.git#d6a97c9476c3377877946873cad040f769e42698",
            .hash = "12204ade9b0ef08b6787667a99396aa1e89e8a6dca1e25b62b9490e3ab8394a0383c" },
    },

    .paths = .{
        "LICENSE",
        "README.md",
        "build.zig",
        "build.zig.zon",
        "src",
    },
}
```
This can obviously not work as I am using the same url for both dependencies. How do I do this correctly?
Also I am a bit confused by the error message from `zig build`:
```
❯ zig build
/Users/jan/projects/open-sensor-platform/firmware/blink/build.zig:10:30: error: root struct of file 'build' has no member named 'init'
    const microzig = MicroZig.init(b, .{});
                     ~~~~~~~~^~~~~
/Users/jan/.cache/zig/p/12204ade9b0ef08b6787667a99396aa1e89e8a6dca1e25b62b9490e3ab8394a0383c/build.zig:1:1: note: struct declared here
const std = @import("std");
^~~~~
referenced by:
    runBuild__anon_8411: /opt/homebrew/Cellar/zig/0.13.0/lib/zig/std/Build.zig:2116:27
    main: /opt/homebrew/Cellar/zig/0.13.0/lib/zig/compiler/build_runner.zig:301:29
    remaining reference traces hidden; use '-freference-trace' to see all reference traces
```
Is this error even linked or completely unrelated?
Feel free to also checkout our repository for the full setup:
https://github.com/opendata-nuernberg/open-sensor-platform/tree/main/firmware/blink

Thanks for your help!

Best,

Jan

Kytezign, 2024-11-14:
I'm new to this zig & this community but I was able to get the RP Pico to blinky in a separate repo.
Not sure it's helpful to you or others but I'll share it here just in case. 

https://github.com/Kytezign/zig_pico_blinky_example

mattnite, 2025-01-18:
Hey @JanBeelte we've overhauled the build API to make it so that you just need to download one dependency. We've also started on building out the website at `https://microzig.tech`, there are some instructions there that you can modify for your specific hardware.

Let me know how it goes.

Yettimania, 2025-03-16:
Hey @mattnite, do the build instructions have to be updated to handle the proper identifiers in `0.14.0`. 

I'm receiving:  

```bash
/home/.cache/zig/tmp/3d29483a57046d0e/build.zig.zon:2:13: error: name must be a valid bare zig identifier (hint: switch from string to enum literal)
    .name = "port/raspberrypi/rp2xxx",
            ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/.cache/zig/tmp/6b3fcbc21d99a241/build.zig.zon:2:13: error: name must be a valid bare zig identifier (hint: switch from string to enum literal)
    .name = "tools/uf2",
            ^~~~~~~~~~~
/home/.cache/zig/tmp/44edf658bccc86a1/build.zig.zon:2:13: error: name must be a valid bare zig identifier (hint: switch from string to enum literal)
    .name = "tools/regz",
            ^~~~~~~~~~~~
/home/.cache/zig/tmp/be57e9bfc7f68b62/build.zig.zon:2:13: error: name must be a valid bare zig identifier (hint: switch from string to enum literal)
    .name = "build/definitions",
            ^~~~~~~~~~~~~~~~~~~
```

Grazfather, 2025-03-16:
If you're using zig 0.14.0, you need to update microzig to the main on or after acd11a0. As far as I can tell, zig 0.14.0 is not backwards compatible at all.

Grazfather, 2025-03-25:
Just ran blinky basically as is today on main. Feel free to ask for more help on the discord.

closed 2025-03-25 by Grazfather