ZigEmbeddedGroup/microzig #313: Regz Outputs LE UTF-16 Text

Bud-ro, 2024-12-11:
Currently when I run `regz` the output is LE UTF-16:
```
$ file generated.zig
generated.zig: Little-endian UTF-16 Unicode text, with CRLF line terminators
```

I'd prefer if the output format was UTF-8 instead of another encoding. Attempting to compile a zig file that is LE UTF-16 encoded results in a very strange error. As it stands I have to re-encode the generated file in order to utilize it in my projects.

Commit: cca83d508e6ea7f024ac95fb74980ff04cfbf554
OS: Windows 10
Version: Zig 0.13.0

Bud-ro, 2024-12-13:
I have since discovered Powershell uses LE UTF-16, and it's very much hard coded :)
So I think the real issue is lacking an option to write the output directly to a file. An option like `-o registers.zig` could allow for UTF-8 to simply be used across every OS.

mattnite, 2025-01-18:
https://github.com/ZigEmbeddedGroup/microzig/blob/cf666d6137b85dfa5353b275c1b46b421f646f3f/tools/regz/src/main.zig#L39

If you run `regz --help` it will tell you the argument for outputting to a file

closed 2025-01-18 by mattnite

Bud-ro, 2025-01-18:
The README suggests directing stdout to the file, without mention of `--output_path` which might be where I got this misconception from... Sorry for making noise on the issue tracker without doing due diligence.