ZigEmbeddedGroup/microzig #370: Add .ARM.extab to generated ARM linker.ld scripts

marnix, 2025-01-31:
Per https://github.com/ziglang/zig/issues/22685#issuecomment-2627678031, Zig 0.14 has enabled 'unwind tables' for ARM embedded (and most other platforms).  Therefore it looks like we need to update `tools/generate_linker_script.zig`, as part of the upgrade to Zig 0.14 (being done on the current [zig-master](https://github.com/ZigEmbeddedGroup/microzig/tree/zig-master) branch).

```diff
diff --git a/tools/generate_linker_script.zig b/tools/generate_linker_script.zig
index 308b055e51..775892c0f6 100644
--- a/tools/generate_linker_script.zig
+++ b/tools/generate_linker_script.zig
@@ -107,6 +107,10 @@
 
         switch (program_args.cpu_arch) {
             .arm, .thumb => try writer.writeAll(
+                \\  .ARM.extab : {
+                \\      *(.ARM.extab* .gnu.linkonce.armextab.*)
+                \\  } >flash0
+                \\  
                 \\  .ARM.exidx : {
                 \\      *(.ARM.exidx* .gnu.linkonce.armexidx.*)
                 \\  } >flash0
```

I have no idea whether the above is actually correct, but it seems to work for me.  (At the time of writing, I did not try to actually debug and look at any stack traces yet...)

I have no idea whether something similar would be needed for RISCV, and/or for other embedded platforms.

marnix, 2025-01-31:
Actually this change can of course already be made on current master, but it's needed for Zig 0.14.

Grazfather, 2025-03-13:
Done in #394 and #373

closed 2025-03-13 by Grazfather