ZigEmbeddedGroup/microzig #376: Consider publishing `.zig.zon` packages to GitHub Releases

ahogen, 2025-02-03:
I see in the readme that this repo can be used as a dependency without cloning the whole thing as a git submodule, by instead writing your own `.zig.zon` file and pointing it at `https://downloads.microzig.tech/` to download release archives.

However, those archives seem to be out of date, the latest there was v0.12.1 but here in GitHub the latest release is 0.13.1. Also the certificate for that downloads site is invalid:

> Websites prove their identity via certificates. Firefox does not trust this site because it uses a certificate that is not valid for `downloads.microzig.tech`. The certificate is only valid for the following names: ackcon.de, ashet.computer, baikal.random-projects.net, booking.shackspace.de, cloud.random-projects.net, club.zig.pm, [....]
>  
> Error code: [SSL_ERROR_BAD_CERT_DOMAIN](about:certerror?e=nssBadCert&u=https%3A//downloads.microzig.tech/&c=UTF-8&d=%20#certificateErrorDebugInformation)

Would you consider adding `.zon` tar files here to GitHub instead?

I'm getting started with an Adafruit Feather M0 and would like to use the `elf2uf2` binary from this repo as a tool in my build.

Thanks!

mattnite, 2025-02-21:
Please go to https://microzig.tech to see the url for the latest tarballs. We changed where we were storing them and didn't realize the old location was mentioned in the readme, thank you for pointing that out!

Grazfather, 2025-03-14:
The readme is acccurate now? We should be able to close this.

mattnite, 2025-03-14:
Yup, I was leaving it open for some breathing time,but it's fixed now.

closed 2025-03-14 by mattnite

ahogen, 2025-03-14:
> MicroZig uses a monorepo architecture, but provides a lot of different packages. If you just want to get started, head over to [downloads.microzig.tech](https://downloads.microzig.tech/) and download an example for the chip family you desire. MicroZig's version listed here match the zig versions assoicated with them, so version 12.0 match zig 12.0. If you don't see the zig version here you will have to clone the repo to get started.

[image]

Grazfather, 2025-03-15:
Ah, yeah it looks like we don't have a cert for that subdomain.

Grazfather, 2025-03-17:
Cert should work again. We still need to update the instructions.