ZigEmbeddedGroup/microzig #389: [RP2040] Regression—blinky doesn't blink with latest microzig

MatthiasPortzel, 2025-02-17:
### Issue
I am no longer able to run programs on my Pico or RP2040 based board with current versions of Microzig. I've reproduced with the blinky example program included in this repo. A bisect points to #373 (5385b0e) as the first "bad" commit. I don't have more time to investigate this issue at present and so will continue using the last working version of microzig (520f370).

### Reproduction
```
git clone https://github.com/ZigEmbeddedGroup/microzig.git
cd microzig
git switch --detach 5385b0e
cd examples/raspberrypi/rp2xxx
zig build
```
Drag `zig-out/firmware/pico_blinky.uf2` onto a Pi Pico

**Expected result**: On-board LED flashes
**Actual result**: On-board LED does not turn on

**Zig version**: 0.13.0

Let me know if anyone is unable to reproduce with a Pico 1 and this example.

MatthiasPortzel, 2025-02-17:
CC @marnix

Grazfather, 2025-02-21:
I can reproduce, but not if I do `zig build --release=fast`. Does this also get around the issue?

marnix, 2025-02-22:
@Grazfather To me it looks like your https://github.com/ZigEmbeddedGroup/microzig/pull/394 is the correct fix for this.

Somehow Zig (0.13.0 and current 0.14-pre) requires an explicit `.ARM.extab` section in a .ld linker script, at least for thumb and/or ARM.  But if one doesn't, the problem is only hit if code is actually generated for that section.

And my https://github.com/ZigEmbeddedGroup/microzig/pull/373 (https://github.com/ZigEmbeddedGroup/microzig/commit/5385b0e2a53ea8dd46e6a4f7990397d1251e4511) did trigger all MicroZig code to start using that section, at least in debug builds, and sometimes also even in `--release=fast` builds.

MatthiasPortzel, 2025-02-23:
#394 does indeed fix this.

I also wanted to mention, with that PR merged into the zig-master branch, I’m able to run my program in Zig 0.14. 🎉

closed 2025-02-23 by MatthiasPortzel