ZigEmbeddedGroup/microzig #420: microzig.Target.chip.cpu_module_file doesn't belong in chip

Grazfather, 2025-03-27:
See discussion [here](https://github.com/ZigEmbeddedGroup/microzig/pull/415#discussion_r2016496198) from #415.

In that PR the esp32-c3 support is expanded to build a binary expected to be loaded at 0 (where the bootrom jumps to it directly) and to another address, where there is a bootloader loads the file, clear out the .bss, etc.

This reveals that this `.cpu_module_file` isn't really a property of the CPU, and probably should be moved up into the Target and renamed to something more generic.

The `derive` method should also be expanded to support overriding this field.

Finally, some targets that make a whole Target from scratch can probably just derive from a similar target (esp32_c3 and esp32_c3_direct_boot being the most obvious).

Grazfather, 2025-04-09:
That would be nice. Probably worth another issue

tact1m4n3, 2025-04-27:
Solved by #450

closed 2025-04-27 by tact1m4n3