ZigEmbeddedGroup/microzig #465: Make the user's app root module of the firmware

tact1m4n3, 2025-04-20:
Currently the root module of the firmware is `core/start.zig`. This has one upside and one downside. The upside is that we can export the startup logic without requiring the user to do anything. However this can be solved by requiring the user to call a function at comptime like this:
```zig
const microzig = @import("microzig");

comptime {
    microzig.cpu.export_startup_logic(); // would export startup logic
}

pub fn main() {

}
```
The downside is that the user loses the ability to control the `root` module, which is often used to configure other dependencies (even std). This has been worked around for `std` but for other deps there is currently no solution that I am aware of.

If the user's module is the root module, then you can specify all the configuration you want directly, like you would do for a normal zig program.

Grazfather, 2025-04-20:
What are some examples of other configurations this would allow the user to setup?

tact1m4n3, 2025-04-22:
Most packages (that have any sort of configuration of course) in zig use this pattern to be configured:
```zig
// root_source_file.zig
pub const {package_name}_options = .{
    ...
};
```
Idk any specific example for embedded, but if the user wanted to use a package like this he would have no way to configure it.

mattnite, 2025-12-27:
Had to open this back up because it had some issues with how I was mixing modules intended for embedded targets and ones for running on the host.

We can essentially get it to the point where we require users to have this boilerplate to their code:

```zig
comptime {
    _ = microzig;
}
```

This forces evaluation of the comptime code which sets up the startup code for our different targets. Note that using microzig in main does not cause evaluation, because main is called by `microzig_main`, which is set up by the startup code, so no matter what, that code block above needs to be evaluated in order for the compiler to look at the application writer's main function.

it should also be noted that this is modeled after how the standard library does things, however, the standard library is special cased in that the root of the standard library is unconditionally evaluated, where we do not have that ability, nor do I think we should add the option to the compiler/build system.