ZigEmbeddedGroup/microzig #684: RP2040 CDC usb example fails for integer overflow

Kytezign, 2025-09-24:
I've followed the getting started instructions then replaced the main.zig with the usb_cdc example here: 

================ STARTING NEW LOGGER ================
[0.506618] info: cdc test: 1
[1.006610] info: cdc test: 2
[1.506611] info: cdc test: 3
[2.006613] info: cdc test: 4
[2.506614] info: cdc test: 5
[3.006615] info: cdc test: 6
[3.009325] error: panic: integer overflow
[3.013256] error: panic: unhandled interrupt

Kytezign, 2025-09-24:
Maybe a duplicate of this:
https://github.com/ZigEmbeddedGroup/microzig/issues/677

If the PR is closed, does that mean it's in the release 
https://github.com/ZigEmbeddedGroup/microzig/releases/download/0.15.0/microzig.tar.gz

Kytezign, 2025-09-25:
I do not yet have a satisfying explanation but after changing the cdc.zig write function to this it's working for me.

Really curious at this point, is this just a me problem somehow? 
```        
pub fn write(self: *@This(), data: []const u8) []const u8 {
            var write_count: usize = 0;
            const w_len = self.tx.get_writable_len() - 1;

            if (data.len >= w_len) {
                self.tx.write_assume_capacity(data[0..w_len]);
                write_count = w_len;
                _ = self.write_flush();
            } else {
                self.tx.write_assume_capacity(data[0..data.len]);
                write_count = data.len;
            }

            return data[write_count..];
        }
```

```
This is very very long text sent from RP Pico by USB CDC to your device: 2
This is very very long text sent from RP Pico by USB CDC to your device: 3
Your message to me was: test 0
This is very very long text sent from RP Pico by USB CDC to your device: 4
This is very very long text sent from RP Pico by USB CDC to your device: 5
Your message to me was: test 1
This is very very long text sent from RP Pico by USB CDC to your device: 6
This is very very long text sent from RP Pico by USB CDC to your device: 7

```

arkadiuszwojcik, 2025-09-25:
@Kytezign I didn't test that code for a long time but for sure there was some changes after zig 0.15 upgrade. Someone implemented custom circular buffer as old one was remove from standard library. But there is also possibility that CDC write had some issue. I will test that when I will find time or maybe you are sure of some bug?

Kytezign, 2025-09-25:
> [@Kytezign](https://github.com/Kytezign) I didn't test that code for a long time but for sure there was some changes after zig 0.15 upgrade. Someone implemented custom circular buffer as old one was [...]

Thank you!  Looks like when I build with ReleaseFast I'm seeing some odd behavior - partial and dropped messages - so whatever I did above isn't a real solution (or I guess there could be another issue).  I'll keep poking around, I'd really like to use zig in my next project.

Kytezign, 2025-09-25:
Another finding: if I increase the length of the buffer  `const FIFO = utilities.CircularBuffer(u8, 256);` the problem goes away until I increase the length of the long string to exceed the buffer length.  
Definitely seems like something is going wrong when that buffer fills up.

Grazfather, 2025-09-25:
you can hopefully debug the circular buffer on your host machine, even in a test block, which would make it a lot faster to iterate on.

Kytezign, 2025-09-26:
At this point - after updating `get_readable_len` - I'm starting to look outside the buffer implementation.  For example if I try to write smaller messages quickly it will hang with flush apparently not doing anything - returning 0 because of device not ready I think.

Kytezign, 2025-10-01:
Just an update, if someone has a similar issue to me - with really large/fast CDC writes.  This version of the write function is really consistent for me.  I'm guessing there is something in the USB implementation that is not waiting for data to actually be sent out before overwriting some buffer.  Here I'm using the SOF_RD count to monitor for USB activity - which is just a stab in the dark on my end.  It seems to work but I'm not sure if it always will.  

I created a pull request for the `get_readable_len` issue: https://github.com/ZigEmbeddedGroup/microzig/pull/697

I'm not sure what to do about this data loss/corruption issue but I'm probably going to call the below good enough for my needs and move onto the my actual project. 

```
// Transfer data to host
var prev_sof_rd_count: u32 = 0;
/// Not 100% sure why this works but the general idea is that we could override the USB buffer if we don't wait long enough
/// the SOF_RD count is an indicator of activity on the bus - hopefully meaning the buffer is empty.
/// surely a better way to do this in the USB implementation...
fn wait_next() void {
    while (prev_sof_rd_count == peripherals.USB.SOF_RD.read().COUNT) {
        usb_dev.task(false) catch unreachable;
    }
    prev_sof_rd_count = peripherals.USB.SOF_RD.read().COUNT;
}

var usb_tx_buff: [1024]u8 = undefined;

pub fn write(comptime fmt: []const u8, args: anytype) void {
    wait_next();
    const text = std.fmt.bufPrint(&usb_tx_buff, fmt, args) catch &.{};
    var write_buff = text;
    while (write_buff.len > 0) {
        write_buff = driver_cdc.write(write_buff);
        wait_next();
    }
    // Short messages are not sent right away; instead, they accumulate in a buffer, so we have to force a flush to send them
    _ = driver_cdc.write_flush();
}
```

Kytezign, 2025-10-01:
Adding a wait for AVAILABLE_0 to de-assert seems to be effective - much better than above - though I'm not sure it's a truly valid solution either but reading the datasheet it _seems_ right:

From: microzig/port/raspberrypi/rp2xxx/src/hal/usb.zig
added `while (ep.buffer_control.?.read().AVAILABLE_0 == 1) {}`
```
        pub fn usb_start_tx(
            ep_addr: u8,
            buffer: []const u8,
        ) void {
            // It is technically possible to support longer buffers but this demo
            // doesn't bother.
            // TODO: assert!(buffer.len() <= 64);
            // You should only be calling this on IN endpoints.
            // TODO: assert!(UsbDir::of_endpoint_addr(ep.descriptor.endpoint_address) == UsbDir::In);

            const ep = hardware_endpoint_get_by_address(ep_addr);
            // wait until the buffer is ready for more data
            while (ep.buffer_control.?.read().AVAILABLE_0 == 1) {}

            // TODO: please fixme: https://github.com/ZigEmbeddedGroup/microzig/issues/452
            std.mem.copyForwards(u8, ep.data_buffer[0..buffer.len], buffer);

            // Configure the IN:
            const np: u1 = if (ep.next_pid_1) 1 else 0;

            // The AVAILABLE bit in the buffer control register should be set
            // separately to the rest of the data in the buffer control register,
            // so that the rest of the data in the buffer control register is
            // accurate when the AVAILABLE bit is set.
```

closed 2025-10-05 by Kytezign